zephyrproject-rtos/ArduinoCore-zephyr #10: Integrate with the Arduino IDE / CLI

beriberikix, 2022-07-04:
Note: this is currently out of scope for the GSoC season due to the time required to implement but I'm leaving this here for future contributions.

One of the key features of the Arduino ecosystem is the integrated development environment, with the most well-known being the classic Java-based IDE (others exist, like a new IDE, Web editor and CLI.) The IDE does many things, including downloading new Cores, new boards and the necessary tools to program and debug boards. When completely new targets are added (like this Zephyr one,) Arduino calls these "platforms" and tools can use a series of metadata files in order to facilitate these capabilities (like fetching new boards.) While there is no [porting guide](https://github.com/arduino/ArduinoCore-API/issues/124,) much of the details are captured in the CLI "[Platform Specification](https://arduino.github.io/arduino-cli/0.24/platform-specification/)" page.

Per the guide there are 3 key .txt files:
* **platform.txt** contains definitions for the CPU architecture used (compiler, build process parameters, tools used for upload, etc.)
* **boards.txt** contains definitions for the boards (board name, parameters for building and uploading sketches, etc.)
* **programmers.txt** contains definitions for external programmers (typically used to burn bootloaders or sketches on a blank CPU/board)

For reference, see the mbed core's [platform.txt](https://github.com/arduino/ArduinoCore-mbed/blob/master/platform.txt), [boards.txt](https://github.com/arduino/ArduinoCore-mbed/blob/master/boards.txt) & [programmers.txt](https://github.com/arduino/ArduinoCore-mbed/blob/master/programmers.txt).

The biggest challenge/unknown is packaging up and distributing the tools needed by zephyr for programming and debugging. For an initial integration would could push the responsibility of installing dependencies on to the user. They would need to place dependencies at well-known locations that would be hardcoded in the platform.txt/programmers.txt. But for a seamless developer experience, we would want the Arduino IDE to automatically download and install all the tools needed by Zephyr. The question is how far do we go? Take a look at the Getting Started guide; there are build system deps like CMake & Device Tree, compilers like the [SDK-NG](https://github.com/zephyrproject-rtos/sdk-ng) and west/pip packages. It's unclear what should be packaged and how. One data point is that the [Nordic nRF Connect for Desktop](https://www.nordicsemi.com/Products/Development-tools/nRF-Connect-for-desktop) does distribute their SDK with all its dependencies, so it is theoretically possible.

DhruvaG2000, 2022-09-25:
@beriberikix I feel like this issue can also be raised in https://github.com/arduino/arduino-ide/issues so that we get attention from the Arduino IDE community. This might help get more people on board or atleast help in getting some direction

beriberikix, 2022-09-25:
Agreed we should figure out how to get more visibility but not sure the right way. Since this is not an actual issue with the IDE, if I was the team I'd immediately close the issue. arduino/ArduinoCore-API#124 seemed like the most logical place but has gotten little visibility. They also don't use GitHub discussion, which would have been my next suggestion. I know they have a [forum](arduino/ArduinoCore-API/issues/124) & [Discord](https://discord.gg/jQJFwW7), but not sure how to use those channels.

At the very least I'll re-ping the original issue.