zephyrproject-rtos/ArduinoCore-zephyr #11: Autogenerate board support

beriberikix, 2022-07-04:
Note: this is currently out of scope for the GSoC season due to the time required to implement but I'm leaving this here for future contributions.

Arduino cores call each board a "[variant](https://arduino.github.io/arduino-cli/0.24/platform-specification/#core-variants)" and place board-specific files in the [`variants` folder](https://github.com/arduino/ArduinoCore-mbed/tree/master/variants). These files, plus specifying them in the boards.txt (see #10,) is what let's the Arduino tools know how to build, flash and generally interact with a board.

In the Zephyr world, a lot of that information is encapsulated in files like a board's [device tree](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/arduino_nano_33_iot/arduino_nano_33_iot.dts). It is theoretically possible to use this information to partially (or fully?) autogenerate Arduino variants and boards.txt. There are many libraries out there, but for example [lopper ](https://github.com/devicetree-org/lopper) is a common Python library to manipulate DTS files.

DhruvaG2000, 2023-10-22:
This is a good idea @beriberikix , and since it involves devicetree I think a potential topic to bring up in coming Linux Plumbers Conf 2023. Do update if you have had any more insights on this topic from since when you've raised this issue.

beriberikix, 2023-10-22:
Okay! I still think creating a custom output for lopper might be the most straightforward. I know the MicroPython folks are asking a similar question.