zephyrproject-rtos/ArduinoCore-zephyr #16: Move board definitions to `variants` folder

beriberikix, 2022-07-10:
Currently the pin mapping for the nano 33 ble is located in the Arduino core itself. Per the Arduino [platform specification](https://arduino.github.io/arduino-cli/0.24/platform-specification/#core-variants), these kind of files should be in their own directory under a `variants` folder (see [mbed](https://github.com/arduino/ArduinoCore-mbed/tree/master/variants) for an example.)

#9 should be updated once complete.

DhruvaG2000, 2022-07-16:
Is this the structure we are aiming for @beriberikix ?

```tree

└─❯ tree . -L 3
.
├── CMakeLists.txt
├── CODE_OF_CONDUCT.md
├── cores
│   ├── arduino
│   │   ├── api -> /home/dhruva/temp/ArduinoCore-API/api
│   │   ├── Arduino.h
│   │   ├── CMakeLists.txt
│   │   ├── main.cpp
│   │   └── zephyrCommon.cpp
│   └── CMakeLists.txt
├── Kconfig
├── LICENSE
├── README.md
├── samples
│   ├── blinky_arduino
│   │   ├── CMakeLists.txt
│   │   ├── prj.conf
│   │   ├── README.rst
│   │   ├── sample.yaml
│   │   └── src
│   ├── button_press_led
│   │   ├── CMakeLists.txt
│   │   ├── prj.conf
│   │   ├── README.rst
│   │   ├── sample.yaml
│   │   └── src
│   └── hello_arduino
│       ├── CMakeLists.txt
│       ├── prj.conf
│       ├── README.rst
│       ├── sample.yaml
│       └── src
├── variants
│   ├── ARDUINO_NANO33BLE
│   │   ├── arduino_nano_33_ble.overlay
│   │   └── arduino_nano_ble_sense_pinmap.h
│   └── CMakeLists.txt
└── zephyr
    └── module.yml
```

beriberikix, 2022-07-16:
Yes, that looks great!

beriberikix, 2022-08-03:
Closed with #17

closed 2022-08-03 by beriberikix