zephyrproject-rtos/ArduinoCore-zephyr #29: Create i2c guide/sample on how to use an Arduino Library with this port

alvarowolfx, 2022-08-08:


DhruvaG2000, 2022-08-28:
@alvarowolfx is there any progress on this? I tried using one of the on-board sensors on the arduino nano 33 ble sense that I have, but I am for some reason unable to get it working:
```
\0\0*** Booting Zephyr OS build zephyr-v3.1.0-4263-g05faddc92388  ***
Started
Failed to initialize IMU!
Accelerometer sample rate = 119.0000 Hz

Acceleration in G's
X       Y       Z
0.3389  1.0000  0.3394
0.3389  1.0000  0.3394
0.3389  1.0000  0.3394
0.3389  1.0000  0.3394
0.3389  1.0000  0.3394
0.3389  1.0000  0.3394
0.3389  1.0000  0.3394
0.3389  1.0000  0.3394
0.3389  1.0000  0.3394
0.3389  1.0000  0.3394
```
These are the logs and it means I2C1 node either doesn't work or there is some bug in our Wire.

The i2c0 connected external ADXL345 worked fine so I find it hard to believe that our Wire.h maybe buggy.
Any comments on this?

DhruvaG2000, 2022-08-28:
I can verify that there is no hardware issue,
```sh
\0*** Booting Zephyr OS build zephyr-v3.1.0-4263-g05faddc92388  ***
Observation:1
Pressure:91.0 kPa
Temperature:27.2 C
Observation:2
Pressure:194.4 kPa
Temperature:27.5 C
Observation:3
Pressure:91.0 kPa
Temperature:27.5 C
Observation:4
Pressure:91.0 kPa
Temperature:27.5 C
Observation:5
Pressure:91.0 kPa
Temperature:27.5 C
Observation:6
Pressure:91.0 kPa
Temperature:27.5 C
```
This is the stock zephyr sample and it definitely works.

DhruvaG2000, 2022-09-25:
I think an i2cdemo sample already exists, + I have also maintained a seperate repo showing ADXL345 Arduino Lib works, hence this can be closed now.

closed 2022-09-25 by DhruvaG2000