zephyrproject-rtos/ArduinoCore-zephyr #38: Serial.print fails

DhruvaG2000, 2022-08-27:
The wrapper for ``Serial.print(val >> i & 1, BIN);`` and ``Serial.print(i, HEX);`` is probably missing which causes build failures.

DhruvaG2000, 2022-08-27:
Refer: https://www.arduino.cc/reference/en/language/functions/communication/serial/print/

> Serial.print(78, BIN) gives "1001110"

Serial.print(78, OCT) gives "116"

Serial.print(78, DEC) gives "78"

Serial.print(78, HEX) gives "4E"

DhruvaG2000, 2022-08-28:
Adding more bugs:
```
[59/162] Building CXX object CMakeFiles/app.dir/src/main.cpp.obj
FAILED: CMakeFiles/app.dir/src/main.cpp.obj
ccache /home/dhruva/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++ -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/dhruva/zephyrproject/zephyr/include -I/home/dhruva/zephyrproject/Arduino_Lib_Demo/build/zephyr/include/generated -I/home/dhruva/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52 -I/home/dhruva/zephyrproject/zephyr/lib/libc/newlib/include -I/home/dhruva/zephyrproject/zephyr/soc/arm/nordic_nrf/common/. -I/home/dhruva/zephyrproject/modules/lib/Arduino-Zephyr-API/cores/arduino/. -I/home/dhruva/zephyrproject/modules/lib/Arduino-Zephyr-API/cores/arduino/../../variants -I/home/dhruva/zephyrproject/modules/lib/Arduino-Zephyr-API/variants/. -I/home/dhruva/zephyrproject/modules/lib/Arduino-Zephyr-API/variants/arduino_nano_33_ble_sense -I/home/dhruva/zephyrproject/modules/lib/Arduino-Zephyr-API/libraries/Wire/. -I/home/dhruva/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/dhruva/zephyrproject/modules/hal/nordic/nrfx -I/home/dhruva/zephyrproject/modules/hal/nordic/nrfx/drivers/include -I/home/dhruva/zephyrproject/modules/hal/nordic/nrfx/mdk -I/home/dhruva/zephyrproject/zephyr/modules/hal_nordic/nrfx/. -I/home/dhruva/zephyrproject/Arduino_Lib_Demo/Arduino_HTS221/. -I/home/dhruva/zephyrproject/Arduino_Lib_Demo/Arduino_HTS221/src -isystem /home/dhruva/zephyrproject/zephyr/subsys/cpp/include -fno-strict-aliasing -Os -fcheck-new -std=c++11 -Wno-register -fno-exceptions -fno-rtti -imacros /home/dhruva/zephyrproject/Arduino_Lib_Demo/build/zephyr/include/generated/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/home/dhruva/zephyr-sdk-0.14.2/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/dhruva/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/dhruva/zephyrproject/Arduino_Lib_Demo=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/dhruva/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/dhruva/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs -nostdinc++ -Wno-unused-variable -Wno-comment -MD -MT CMakeFiles/app.dir/src/main.cpp.obj -MF CMakeFiles/app.dir/src/main.cpp.obj.d -o CMakeFiles/app.dir/src/main.cpp.obj -c /home/dhruva/zephyrproject/Arduino_Lib_Demo/src/main.cpp
/home/dhruva/zephyrproject/Arduino_Lib_Demo/src/main.cpp: In function 'void setup()':
/home/dhruva/zephyrproject/Arduino_Lib_Demo/src/main.cpp:23:10: error: no match for 'operator!' (operand type is 'arduino::ZephyrSerial')
   23 |   while (!Serial);
      |          ^~~~~~~
/home/dhruva/zephyrproject/Arduino_Lib_Demo/src/main.cpp:23:10: note: candidate: 'operator!(bool)' (built-in)
/home/dhruva/zephyrproject/Arduino_Lib_Demo/src/main.cpp:23:10: note:   no known conversion for argument 1 from 'arduino::ZephyrSerial' to 'bool'
[68/162] Building CXX object zephyr/CMakeFiles/zephyr.dir/home/dhruva/zephyrproject/modules/lib/Arduino-Zephyr-API/cores/arduino/main.cpp.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/dhruva/zephyrproject/Arduino_Lib_Demo/build
[ble: exit 1]
```

DhruvaG2000, 2022-08-28:
Also need to add ``#include <zephyrSerial.h>`` to Arduino.h or else it doesn't find Serial

soburi, 2022-12-12:
@DhruvaG2000 
This issue may solve with #64.
Could you confirm it?

DhruvaG2000, 2022-12-17:
Yes now we need to clean up the ``samples/hello_arduino``. we can delete the ``#include "zephyrSerial.h"`` line and also cleanup a bit.

DhruvaG2000, 2023-07-25:
closing this issue as per https://github.com/zephyrproject-rtos/gsoc-2022-arduino-core/pull/71#issuecomment-1509495736

closed 2023-07-25 by DhruvaG2000