zephyrproject-rtos/ArduinoCore-zephyr #87: Implementing SPI library

soburi, 2023-10-22:
We need to implement the SPI library to keep it compatible with standard AVR Arduino.

DhruvaG2000, 2023-10-22:
Just documenting further details here,
1. [SPI.h arduino API](https://www.arduino.cc/reference/en/language/functions/communication/spi/)
2. [Zephyr SPI API](https://docs.zephyrproject.org/apidoc/latest/group__spi__interface.html)

DhruvaG2000, 2024-05-21:
> SPI should be included when writing a device overlay, but I'm not in a position to test it.

@soburi  I could probably test the SPI implementation if you're able to raise a PR for SPI using ADXL355 sensor or something that I may have access to.

https://github.com/plasmapper/adxl355-arduino
This should be a good library to validate SPI works. Let me know if you are interested, I can definitely help but will be difficult for me to implement fully at the moment.

closed 2024-06-26 by DhruvaG2000