zephyrproject-rtos/ArduinoCore-zephyr #88: Implementing EEPROM library

soburi, 2023-10-22:
We must implement the EEPROM library to keep it compatible with standard AVR Arduino.
Where the EEPROM library implementation does not even have actual EEPROM, it should work with various non-volatile memory such as flash to keep compatibility.

DhruvaG2000, 2023-10-22:
@soburi  Do you syggest using something like https://docs.zephyrproject.org/latest/services/storage/nvs/nvs.html underneath?

So basically a wrapper of all funcs [here](https://docs.arduino.cc/learn/built-in-libraries/eeprom) for ``nvs_write``, ``nvs_read``, etc. in zephyr?

soburi, 2023-10-22:
> So basically a wrapper of all funcs here for nvs_write, nvs_read, etc. in zephyr?

Yes, it is. The importance is easily migrating from Arduino code here.
In practice, some ArduinoCore implements the EEPROM function using flash memory.

purviyeshi, 2024-09-03:
@DhruvaG2000 I would like to work on this issue, if it is not currently assigned to anyone, could you please assign it to me?

soburi, 2024-09-04:
@purviyeshi 

> @DhruvaG2000 I would like to work on this issue, if it is not currently assigned to anyone, could you please assign it to me?

Anyone can take a part of this problem.
For now, We haven't been giving direction to anyone.
If You submit a PR, even a draft, you will naturally be considered the person in charge.

purviyeshi, 2024-09-04:
> Anyone can take a part of this problem.
> For now, We haven't been giving direction to anyone.
> If You submit a PR, even a draft, you will naturally be considered the person in charge.

Okay, Thank you.
I'll start work.