zephyrproject-rtos/ArduinoCore-zephyr #123: Arduino moving to Zephyr

powerfeatherdev, 2024-09-20:
This is not really an issue, but rather a question and I haven't found any other channel to ask it.

With ARM discontinuing mbedOS support, Arduino has announced their own effort in using Zephyr as a base: https://blog.arduino.cc/2024/07/24/the-end-of-mbed-marks-a-new-beginning-for-arduino/

How would that affect this project?

soburi, 2024-09-21:
Hello, @powerfeatherdev 

I think it is good to reference the discussion in discord.

https://discord.com/channels/720317445772017664/1021855270982402058/1267894806986620928

I don't know the specific policy yet, but I understand that this is also something that is being considered as part of the implementation.

powerfeatherdev, 2024-09-22:
Ah, was not aware that there is a Discord server. Will close this issue and just follow the developments there. Thank you!

closed 2024-09-22 by powerfeatherdev